atsamd-rs/atsamd #744: CI falures for tier >1 BSPs shouldn't interfere with HAL work

ianrrees, 2024-08-18:
Our current CI setup builds all the BSP examples, and a failure in one results in subsequent builds not being run.  This creates a problem when a BSP failure is caused by some change outside of our control, for instance a dependency breaking SemVer conventions in an update.

This is particularly egregious when a tier >1 BSP build fails, for a CI run meant to test changes to the HAL:

[image]

One possible solution is to do the Tier 1 BSP builds before starting on the lower-tier BSPs.

ianrrees, 2024-08-18:
I've made a naive attempt at fixing this by changing the list-BSPs action to provide the BSPs ordered by tier - but the build-BSPs workflow seems to do the builds in random order anyway.

Two possible next approaches spring to mind:

1. See if the jobs matrix can work with JSON objects, or just strings.  If so, instead of supplying bsps as `{"bsp":["arduino_mkr1000","arduino_mkrvidor4000"...`, use something that includes the tier like `{"bsp":[{"name": "arduino_mkr1000", "tier": 2},{"name": "arduino_mkrvidor4000","tier": 2}...`.  I believe this would allow the build-bsp workflow to allow `continue-on-error` to reference the tier.
2. Split the build-bsp workflow in to two, for tier 1 and >1 BSPs.

ianrrees, 2024-08-18:
There are a few subtly different ways to have the BSP build jobs continue after a failure, which to use basically comes down to whether we want the workflow to show a  ❌ (set [`fail-fast`](https://docs.github.com/en/actions/writing-workflows/workflow-syntax-for-github-actions#jobsjob_idstrategyfail-fast) to false) or ✔️ (use a tier clause in `continue-on-error`) depending on whether _any_ BSP builds fail.  We currently use `continue-on-error` for builds with Rust nightly, so get a ✔️ despite all the nightly builds failing clippy.

Some experimentation and discussion in Matrix shows that the success or failure of the CI workflow doesn't directly control whether a PR can be merged (without ticking the box with the big warning ;) ).  Currently, the required checks are set through the project's branch protection settings - for the BSPs, this is a manually curated list of the Tier 1 builds with stable Rust.

The prototype I've PRed initially used `continue-on-error` which required a little faffing around with JSON to pass the tier information to the build matrix.  We could consider using that information to make a job to run after all the builds have finished, and error out if any tier 1 stable builds failed.  This would allow us to simplify the branch protection rules (to just depend on this new job), so the project settings would not have to change if we add/remove tier 1 BSPs.

ianrrees, 2024-08-19:
I've implemented that last paragraph, it's quite nice IMHO.  Demo is available here https://github.com/ianrrees/atsamd/pull/3 - note the only check that's required through GitHub settings is "Tier 1 BSPs all built on stable" so the crates.json file is the single source of truth about which BSPs are required to build.

jbeaurivage, 2024-08-20:
Closed by #745

closed 2024-08-20 by jbeaurivage