atsamd-rs/atsamd #784: Remove deprecated code - embedded-hal 0.2

jbeaurivage, 2024-11-13:
This is a tracking issue for the removal of deprecated code.

- [ ] Remove support for `embedded-hal` v0.2
- [ ] Remove deprecated items in `sercom::dma`
- [ ] Optionnally, when #635 lands, remove the `spi_future` module - Would appreciate @bradleyharden's input on this.

Sympatron, 2024-11-14:
I wouldn't remove `embedded-hal` v0.2 just yet. It can still be useful. Maybe hiding it behind a feature for now would be enogh.

jbeaurivage, 2024-11-14:
Fine by me - out of curiosity, what would you consider a good timeline to start removing it?

Sympatron, 2024-11-14:
I just looked into it further and I think it's not as bad as I thought. I knew some things were missing in v1, because they couldn't settle on an API. But I think they were sufficiently useless in their current state that they probably weren't used anyway. 

Some drivers might still use v0.2, but those could be easily adapted (if they are still maintained...) or you could use [`embedded_hal_compat`](https://docs.rs/embedded-hal-compat/latest/embedded_hal_compat/).

So it's probably safe to remove v0.2 with the next minor version bump.