atsamd-rs/atsamd #810: Provide defmt feature in BSPs to enable atsamd-hal/defmt

ianrrees, 2025-01-10:
In a firmware that depends on a BSP, it's a bit awkward to use defmt for errors that originate in the HAL.  One needs to add the same version of the HAL as the BSP uses to Cargo.toml, to enable the defmt feature.

To resolve this, we should have all the BSPs (at least, that use a new-enough HAL) support a defmt feature themselves.

sajattack, 2025-01-10:
I think part of the difficulty here is that defmt requires an addition to the linker script?

ianrrees, 2025-01-10:
I'm not sure I follow - yes using defmt in a firmware requires modifying the linker script, but even with that in place it's not straightforward to use defmt in a firmware that depends on a BSP directly.

For example, if my firmware's Cargo.toml has something like this:

```toml
[dependencies]
metro_m4 = { version = "0.16.1", features = ["usb"] }
```

and in my firmware, I want to use something like:
```Rust
#[derive(Error, defmt::Format, Debug)]
enum Error {
    #[error("UART error:")]
    UartError(hal::sercom::uart::UartError),
   // ...more variants..
}
```

Currently, the only way AFAICT to do this, is to modify Cargo.toml to add a dependency on the HAL, using the same version as the BSP:
```toml
[dependencies.atsamd-hal]
version = "0.20.2"
default-features = false
features = ["defmt"]
```

What I'm proposing is that it would be nice to be able to just do:
```toml
[dependencies]
metro_m4 = { version = "0.16.1", features = ["usb", "defmt"] }
```

sajattack, 2025-01-10:
Oh, yeah, I know, I'm just saying adding the dep to the default linker script might be slightly annoying and I don't think it can be feature gated. But it's probably not an issue. Just something to consider.

ianrrees, 2025-01-10:
Ah, you're a step ahead of me: if the feature is enabled, but the linker script isn't modified, then the build fails.  This means that the `--all-features` builds as are done by CI would require us to modify the default linker script if we added this defmt feature to BSPs...

So yes, probably deserves a bit more thought, I'll remove the "good first issue" tag!

ianrrees, 2025-03-16:
I stumbled on [this](https://doc.rust-lang.org/cargo/reference/build-scripts.html#rustc-link-arg-bins) today - looks feasible to use a build script to set the linker flag when the defmt feature is selected.  By default, build scripts run when features change.