atsamd-rs/atsamd #819: `clock::v2`: DPLL bug, potential UB?

jbeaurivage, 2025-02-07:
I've been trying out the `clock::v2` module, and I've encountered some pretty strange behavior when trying to setup the DPLL to speed up the main CPU clock.

## My test setup

* Metro M4 (SAMD51)
* The following example code:
```rs
#![no_std]
#![no_main]

use defmt_rtt as _;
use panic_probe as _;

use atsamd_hal::{
    clock::v2::{
        clock_system_at_reset,
        dpll::Dpll,
        gclk::{Gclk, GclkDiv16},
        pclk::Pclk,
    },
    pac::Peripherals,
};

#[cortex_m_rt::entry]
fn main() -> ! {
    let mut peripherals = Peripherals::take().unwrap();

    let (_buses, clocks, tokens) = clock_system_at_reset(
        peripherals.oscctrl,
        peripherals.osc32kctrl,
        peripherals.gclk,
        peripherals.mclk,
        &mut peripherals.nvmctrl,
    );

    // Set GCLK1 to use the 48 MHz DFLL...
    let (gclk1, dfll) = Gclk::from_source(tokens.gclks.gclk1, clocks.dfll);
    // ...and divide it down to 2 MHz
    let gclk1 = gclk1.div(GclkDiv16::Div(24)).enable();
    // Feed GCLK1 to the DPLL
    let (pclk_dpll0, _gclk1) = Pclk::enable(tokens.pclks.dpll0, gclk1);
    // Now start the DPLL with a 50x multiplier, to get a 100 MHz clock,
    // which is the maximum speed the ADC may run at.
    let dpll0 = Dpll::from_pclk(tokens.dpll0, pclk_dpll0)
        .loop_div(50, 0)
        .enable();

    defmt::info!(
        "DPLL: Ready: {}, Locked: {}",
        dpll0.is_ready(),
        dpll0.is_locked()
    );

    // When the DPLL is ready, swap the GCLK0 source to run the CPU
    // on the 100 MHz DPLL
    while !(dpll0.is_ready() && dpll0.is_locked()) {}
    let (_gclk0, _dfll, dpll0) = clocks.gclk0.swap_sources(dfll, dpll0);

    defmt::info!(
        "DPLL: Ready: {}, Locked: {}",
        dpll0.is_ready(),
        dpll0.is_locked()
    );

    loop {
        // A long `nop` delay as a rough way of comparing CPU clock speeds
        defmt::debug!("Tick");
        cortex_m::asm::delay(100_000_000);
    }
}
```

Execution always stalls at this part:
```rs
while !(dpll0.is_ready() && dpll0.is_locked()) {
        core::hint::spin_loop();
}
```
which is making me believe there might be a missed step in setting up the clock tree. What's weird, is that if I move the wait to *after* the call to `swap_sources`, then the CLKRDY and LOCK bits are set correctly, and execution resumes.

What's even weirder however, is that **seemingly random changes in the code affect whether or not the CPU clock is actually sped up** to 100 MHz. For example, adding the following line before `main` makes the clock run at the expected speed:

```rs
rtc_monotonic!(Mono, rtc_clock::Clock32k);
```

But when it's removed, the CPU clock seems to stay stuck at 48 MHz, which is the speed at reset. However that macro is just a bunch of struct and function declarations, and should have no effect on runtime behavior... Leading me to believe that there must be some kind of undefined behavior or misoptimization lurking somewhere. 

I haven't dove deeper into the issue yet, but I wanted to document my findings while it's still fresh.

jbeaurivage, 2025-02-11:
Potentially another report here: https://github.com/atsamd-rs/atsamd/issues/809#issuecomment-2579589129

ianrrees, 2025-04-03:
I've started poking at this, starting with the apparent difference in whether the CPU clock is sped up.  It is weird indeed, but I suspect it boils down to a need to configure filters in the DPLL.

With the hacked-up copy of the above, I'm actually able to change between the fast and slow conditions just by enabling the rtic and/or usb features of the Metro M4 BSP.  I haven't looked in to the RTIC stuff at all, but presume the USB could cause initialization of some global structures, affecting timing of the power up sequence/state?  Anyway...

Looking at the registers for (in PyCortexMDebug notation) GCLK GENCTRL[0], GCLK GENCTRL[1], OSCCTRL, OSCCTRL DPLL[0], and MCLK - there are no differences between the fast and slow case when execution has gotten to to the main loop.

So, as a bit of a sanity check (wondering if somehow the "slow" case was not 48MHz default from the DFLL, but instead the DPLL was locking to half of 2MHz (perhaps it does some sort of clever oversampling, or maybe that's something digital PLLs can do?) to get 50MHz, I decided to bring GCLK0 out to a GPIO pin and measure it.

Both the fast and slow cases -as indicated by the rate of updates of defmt output- produce 100MHz at PB14 (D5 on Metro M4).  However, in the slow case, the clock is unstable - on my oscilloscope I noticed a bit of fuzz on the display a few cycles from the trigger, however on my VNA it's much more obvious.

Sorry about the poor screenshot - I can't remember where I put the code that grabs a screenshot over ethernet, and it doesn't want to read the one thumb drive I found...  First, the fast condition looks as expected:

[image]

The slow condition; I think the chip was correct in indicating the DPLL0 wasn't in a state to use:

[image]

ianrrees, 2025-04-03:
Hmm, this morning I tried the same GPIO approach but with the GCLK1 output which should be 2MHz, and observed it has some odd behaviour in the slow case too - could be the issue is further back in the chain, maybe an issue with the DFLL reference, maybe a measurement artifact?

I need to study the datasheet and errata a bit closer, but have some IRL stuff to do today so progress may be slower than usual.

ianrrees, 2025-04-05:
Interesting data point: I tried making a project in START, but its clock configuration complains that the DFLL48M "Input is Disabled" when attempting the same configuration as the example code uses.  In the DFLL48M block, I've got open loop mode selected.  START is pretty funky, so I don't think this necessarily indicates a problem however it does make me wonder if we're missing something about the DFLL48M configuration.

[image]

ianrrees, 2025-04-05:
> Execution always stalls at this part:
> 
> ```Rust
> while !(dpll0.is_ready() && dpll0.is_locked()) {
>         core::hint::spin_loop();
> }
> ```
> 
> which is making me believe there might be a missed step in setting up the clock tree

Yes, the issue there is that DPLL0 has the ONDEMAND bit set, and it's not running because no peripheral is requesting it.  That issue is resolved like this:

```Rust
    let dpll0 = Dpll::from_pclk(tokens.dpll0, pclk_dpll0)
        .loop_div(50, 0)
        .on_demand(false)
        .enable();
```

However, this still leaves the instability issue - just enabling the `rtic` feature causes the firmware to sometimes run considerably slower.  My theory is that the combination of the open-loop DFLL48M and the DPLL0 filter settings is somehow unstable.

Thinking I'll try two things:

1. Extend the API to allow adjusting the DPLL filter settings.  The datasheet says about the FILTER settings (as opposed to DCOFILTER) "The DPLL digital filter coefficients are automatically adjusted in order to provide a good compromise between stability and jitter" however it doesn't explain how that works nor how to pick good values...
2. Try using DFLL in closed loop mode and see if anything changes.

ianrrees, 2025-04-05:
Hmm, fiddling with the spectrum analyzer a bit more, I'm finding that with the DPLL0 code completely removed, the characteristics of the DFLL48M (through GCLK1, output via PB15) change noticeably depending on the enabled features.  So, I guess that's a little closer to a minimal reproduction of some quite strange behaviour.  Note the spike at 3.7MHz, this would be the "slow" case:

[image]

This would be the "fast" case:

[image]

ianrrees, 2025-04-05:
As discussed on Matrix, I've implemented the ability to set DPLL PI filter coefficients.  Work in progress at [my issue-819 branch](https://github.com/ianrrees/atsamd/tree/issue-819).

With the test setup I've been using for this bug, I'm not noticing a difference with filter settings, however just the act of setting the variable in the DPLL `Settings` struct (so, just changing the timing ever so slightly) does have an observable effect on the DPLL output spectrum - I presume this effect is rooted in the DFLL48M but haven't confirmed.

Need to do some IRL stuff, but thinking I'll continue by adding a DCO filter setting just for completeness then try to confirm the new settings are doing what they're supposed to.

ianrrees, 2025-04-06:
Random thought about DPLL0 not starting up due to lack of a clock request: I wonder if we should change `Dpll::new()` to default `on_demand: false` so that the maybe-surprising behaviour is opt-in.

ianrrees, 2025-04-10:
I've set up an experiment that I think shows the issue is not in the DFLL48M.  With the clock system like this, I could reproduce the same behaviour where toggling the `usb` feature makes the defmt output go much slower (haven't got GCLK0 going to spectrum analyzer at the moment, would need to cut up another coax cable):

```text
                                        cable
DFLL48M -> GCLK1 (/24) -> PB15 aka D6 >---------> PB16 aka D3 -> GCLK2 -> DPLL0 (x50) -> GCLK0
```

Then, with this arrangement, I can still reproduce the same behaviour:

```text
Signal generator (2MHz square wave) >----------\
                                                \
DFLL48M -> GCLK1 (/24) -> PB15 aka D6 >--x       \---> PB16 aka D3 -> GCLK2 -> DPLL0 (x50) -> GCLK0
```

ianrrees, 2025-04-11:
Changing the NVMCTRL flash read wait state configuration from auto to a hardcoded 6 wait states, doesn't resolve the weird behaviour.

Also changed the main loop slightly, to twiddle a GPIO pin and confirmed that the behaviour continues with debugger disconnected.  That test also indicated that the amount of time the `cortex_m::asm::delay()` calls took was changing in the slow/fast conditions (it wasn't something like defmt logging stalling)

Next thing I'm thinking is to decompile and examine a "slow" and "fast" firmware to see if there's some noticeable difference, maybe in `delay()`.

ianrrees, 2025-04-12:
Compilation of the busy loop looks practically identical, and very reasonable, between the fast and slow firmwares.

ianrrees, 2025-04-13:
Ok, I'm basically convinced that this is not a bug about our clocking code.  Have also confirmed that the same thing happens on a PyGamer (same micro, different layout).  With this example code on a Metro M4:

```Rust
#![no_std]
#![no_main]

use metro_m4 as bsp;

use defmt_rtt as _;
use panic_probe as _;
use bsp::hal::pac::Peripherals;
use bsp::ehal::digital::OutputPin;
use cortex_m::interrupt;

#[cortex_m_rt::entry]
fn main() -> ! {
    interrupt::disable();
    let peripherals = Peripherals::take().unwrap();
    let pins = bsp::Pins::new(peripherals.port);
    let mut debug_pin: bsp::RedLed = pins.d13.into();

    defmt::debug!("Fast mode enabled");
    loop {
        cortex_m::asm::delay(100_000_000);
        debug_pin.set_high().unwrap();
        debug_pin.set_low().unwrap();
        defmt::debug!("Tick");
    }
}
```

I see a pulse every 3.132 seconds, and if the `Fast mode enabled` line is commented out, a pulse every 7.319 seconds.  At 48MHz, 100 million clock ticks should take 2.083 seconds, so even the fast case isn't quite right.

Expanding on my last comment here, in both the fast and the slow case the delay is compiled to this (in different locations, but same instructions):
```
                             **************************************************************
                             * DWARF original prototype: void __delay(u32 cyc)            *
                             **************************************************************
                             void __rustcall __delay(u32 cyc)
                               assume LRset = 0x0
                               assume TMode = 0x1
             void              <VOID>         <RETURN>
             u32               r0:4           cyc
                             __delay                                         XREF[3]:     Entry Point(*), 
                             lib::__delay                                                 delay:0000473e(c), 
                                                                                          .debug_frame::00003464(*)  
        00006938 01 21           movs       r1,#0x1
        0000693a 01 eb 50 00     add.w      cyc,r1,cyc, lsr #0x1
                             LAB_0000693e                                    XREF[1]:     00006940(j)  
        0000693e 01 38           subs       cyc,#0x1
        00006940 fd d1           bne        LAB_0000693e
        00006942 70 47           bx         lr

```

My Ghidra skills are pretty basic; I've had a casual look for any differences between the two programs but keep getting lost in the weeds.

ianrrees, 2025-04-13:
I've stripped this down further by removing defmt and still can reproduce the issue with a `--release` build.  In the `loop{}` are two ways to get a delay of 100M counts, toggle between them to get the fast and slow cases:

```Rust
#![no_std]
#![no_main]

use metro_m4 as bsp;

use panic_halt as _;
use bsp::hal::pac::Peripherals;
use bsp::ehal::digital::OutputPin;
use cortex_m::interrupt;

#[cortex_m_rt::entry]
fn main() -> ! {
    interrupt::disable();
    let peripherals = Peripherals::take().unwrap();
    let pins = bsp::Pins::new(peripherals.port);
    let mut debug_pin: bsp::RedLed = pins.d13.into();

    loop {
        // use this delay for the slow case
        // cortex_m::asm::delay(100_000_000);

        // use this delay for the fast case
        for _ in 0..10 {
            cortex_m::asm::delay(10_000_000);
        }
        debug_pin.set_high().unwrap();
        debug_pin.set_low().unwrap();
    }
}
```

ianrrees, 2025-04-13:
Aha, got it: examining the implementation of [`delay()`](https://github.com/rust-embedded/cortex-m/blob/898b11c7783246fe0137537d20898b91e58ff6aa/cortex-m/src/asm.rs#L17-L44), and the [Cortex-M4 TRM](https://developer.arm.com/documentation/ddi0439/b/Programmers-Model/Instruction-set-summary/Cortex-M4-instructions) we can see that the `bne` will usually take 2-4 cycles depending on the alignment of the target instruction.  That alignment depends on the vagaries of compilation.

This can be demonstrated by making a few copies of `delay()` in the firmware and unrolling the loop to use them in rotation.  Some will be fast, some will be slow.

That also explains why the "fast" case timing isn't quite right.

ianrrees, 2025-04-14:
I've opened a [PR in cortex-m](https://github.com/rust-embedded/cortex-m/pull/587) to resolve the variation in `delay()`, so closing this out.

closed 2025-04-14 by ianrrees