atsamd-rs/atsamd #830: Consolidating BSPs and/or their examples

ianrrees, 2025-03-16:
As we gain more BSPs and examples in them, the maintenance burden is growing and my feeling is that this distracts from HAL development which is probably more valuable.  This issue explores possible options to improve the situation.

## Project management program
Move the examples from BSPs to a separate directory and transform them in to templates, so that eg `use metro_m0 as bsp;` becomes generic `use {{bsp}} as bsp;`, and add a utility that can render these templates out to the BSPs.  This is prototyped in https://github.com/ianrrees/atsamd/tree/manage

### Pro
 * Conceptually simple
 * Easy to integrate in to existing CI
 * No change from the perspective of a BSP end-user
   - It remains clear which examples are available for which board

### Challenges
 * The utility will need to update BSPs' Cargo.toml to meet examples' dependencies

### Con
 * Bespoke code generation makes the overall project less approachable, seems a bit dirty

## Crate per-example
Rather than providing examples using Cargo's functionality, provide them as stand-alone crates which select a target board using a Cargo feature.  To improve developer ergonomics, we probably want to transform/consolidate the BSPs (for each target chip?), so that the BSP provides little more than the chip-board pin mappings.  Doing this would remove most of the `#[cfg()]` logic from the example source code.

`build.rs` can be leveraged to ensure that exactly one board is selected, and populate the correct `memory.x` and `.cargo/config.toml` files.

This is prototyped in https://github.com/ianrrees/atsamd/tree/crate-per-example

### Pro
 * Less Cargo friction to turn an example in to a custom firmware
 * It's clear which dependencies are required by any particular example, as they're not all combined in to a single `[dev-dependencies]` section in a BSP.

### Challenges
 * How to handle memory.x and .cargo/config.toml across different core+memory?
   - build.rs is one option
 * CI needs a way to iterate over supported target boards

### Con
 * Would be a big user-facing change

ianrrees, 2025-03-16:
@bradleyharden - following on from Matrix chat yesterday, I'm tinkering with the crate-per-example approach this morning so will probably edit the description until a discussion gets started.

ianrrees, 2025-03-17:
Just out of interest, we've currently got 35 BSPs for 10 different chips, roughly 225 examples with 93 unique names.

ianrrees, 2025-03-17:
In Matrix chat, there's been a preference for using facilities of Rust/Cargo, rather than a new code generation tool (described above as a project management program).  From my exploration so far, an apparent crux is in the example source code's inclusion of the BSP via:

```Rust
use metro_m0 as bsp;
```

If we want to use the same example source file in multiple BSP crates, this is tricky because there doesn't seem to be a way to rename the BSP in its own Cargo.toml .  One slightly sneaky option is to add a file `examples/bsp.rs` in the BSP with contents like:

```Rust
//! Alias metro_m0 BSP to a generically named bsp, for re-use of example source
pub use metro_m0::*;
```

Then, in the example, we can:
```diff
diff --git a/boards/metro_m0/examples/usb_echo.rs b/boards/metro_m0/examples/usb_echo.rs
index 04d7e4984..2b4e5b77b 100644
--- a/boards/metro_m0/examples/usb_echo.rs
+++ b/boards/metro_m0/examples/usb_echo.rs
@@ -11,7 +11,7 @@ use usbd_serial::{SerialPort, USB_CLASS_CDC};
 
 use bsp::hal;
 use bsp::pac;
-use metro_m0 as bsp;
+mod bsp;
 
 use bsp::entry;
 use hal::clock::GenericClockController;

```

Sympatron, 2025-03-18:
I think this would be great.
There should be a comment stating that this line has to be replaced with `use <name_of_bsp> as bsp;` if used outside the examples directory.

ianrrees, 2025-03-23:
I've updated the top comment here again to add the symlinks section, that option is currently my favourite.

Would be good to get some feedback from people who use Windows, my understanding is that should work OK but I've not got a Windows machine handy to test myself.

ianrrees, 2025-03-28:
Having attempted to make the blinky_basic example generic across a few BSPs, I'm struck by the scale of the challenge we've created...  There are just a couple pairs of BSPs that can use the same blinky_basic source (aside from the BSP name), without modifying the BSP.  Several smaller differences in some boards might not be feasible to make generic (eg some pet the watchdog, one has a comment that an LED and resistor need to be added, one uses neopixel so probably shouldn't be called blinky_basic).

Not saying I think we should abandon this idea, but perhaps we need to think a bit harder about how/whether to add BSPs so that we minimize differences between examples.

9names, 2025-03-28:
When we took on BSPs in `rp-rs` (following in `atsamd-rs` footsteps), the original stance wrt examples was that anything that didn't require external hardware went in the HAL, anything board agnostic went into the `rp-pico` bsp, anything else went to the specific BSP, and every BSP got 1 "showcase" example that showed off everything at ones.
Even with these restrictions, I don't think board examples (or BSPs in general) have paid for the amount of investment in them.
If BSPs are supposed to provide a decent board-level abstraction, you should be able to use the same example across all of them *without changing the example*.
If you can't easily do that in the project, your users also have that problem, and that's not great.

ianrrees, 2025-03-31:
I tend to agree with that sentiment @9names ; the current model doesn't seem sustainable, and it doesn't seem like a quick change is feasible.  My feeling is that the BSPs are useful and probably make sense to have associated with the same project as the HAL, and many/most of the examples are useful in concept.  What's missing is a structure/standard that keeps the maintenance burden from expanding too quickly, which is what this issue is about.

As you suggest, cases where examples aren't portable across different BSPs often indicate flaws in the HAL abstractions [0].  I think part of the problem is that when we've made improvements to the HAL, the examples simply haven't been updated coherently - I'm certainly guilty of making minimal changes to examples when updating BSPs to use more recent HALs!

It seems to me that the ability of a structure/standard for BSPs and their examples, might be judged partly on how well it can handle changes to the HAL API.

[0]: I say often, because I think there are some differences, like whether a board's supplied bootloader enables the watchdog, which are more-or-less out of scope in the current model.  Another example is how the smaller/bigger chips require different ISRs to do interrupt-driven USB.

jbeaurivage, 2025-06-20:
@ianrrees, I'm currently looking at [your `manage` branch](https://github.com/ianrrees/atsamd/tree/manage). If I understand correctly, all the examples would live in the `examples` directory? Right now it seems like all examples from all BSPs have been copied in there, I'm assuming as a way to group them all together and start chipping away. 

Would it make sense to keep a manifest of all examples in there as well (either in toml, yaml or whatever), to give the management tool context wrt board affinities/required features? In time maybe this could be used to render a nicely formatted markdown document with a list of examples, helping users navigate them.

ianrrees, 2025-06-20:
Yep, that's exactly correct.  I implemented a slightly hacky version of the manifest, which we can totally change if desired; the filenames in the examples directory can be prefixed with either a BSP name to indicate that they're specific to one BSP, or `generic` to indicate that the `manage` program should read a config file `examples/examples.toml` to know where to distribute that example.

Sorry, I'm just realising now that I didn't follow through with showing a generic example in there - I started with blinky_basic but went down a rabbit hole looking at how much variance there is between those...  Can do that soon.

ianrrees, 2025-06-20:
I've updated that branch to show a generic (across two whole BSPs!) blinky_basic and rebase it on to current master.  Heading out for a couple days mostly AFK, so I'll be slow to follow up.

ianrrees, 2025-06-20:
If we go forward with this, we'll want to add a CI check that running `manage example distribute` doesn't change the rendered examples

ianrrees, 2025-06-27:
@rnd-ash - I've updated the top-level comment to expand on quantity vs approach.

The main problem I see with the crate per-example approach is that we have a whole lot of different examples (see [above comment](https://github.com/atsamd-rs/atsamd/issues/830#issuecomment-2727742262)).  Unless we deleted nearly all the examples, it would take a substantial amount of work before we had less support burden (which I'll claim is roughly proportional to lines of deduplicated code, including Cargo.toml etc) than we currently do.  That's obviously mixing the quantity and approach questions, but if we hold quantity constant, it's not clear to me that crate per-example would be an improvement.  If we condensed the 35 BSPs down to the 10 different chips they use, we'd lose 25 Cargo.toml files, but we'd gain a Cargo.toml for every unique example, roughly 100 at present.  Of course some of those 100-odd examples aren't unique, but nowhere near 75% of them could be deduplicated.

Most of the work in implementing crate per-example would look very similar to the templated example approach; either way, we'll mostly be smoothing out differences between examples and BSPs.

Both approaches require a bit of initial setup work: making the management tool, or making a pattern to support one example across multiple BSPs.  Beyond that, from the maintainer perspective it's basically about picking an example and adapting it (and the BSPs) to one scheme or the other.  From the crate user's perspective though, this transitional period would be less nice in the crate per-example scenario, because for any particular BSP there will be some examples in the old way and some in the new way, and any example could be in a BSP crate or in its own crate.  With the templated example approach, there is no difference from the crate user's perspective.

closed 2025-07-12 by ianrrees