atsamd-rs/atsamd #858: I2C transaction continued read broken

fko-kuptec, 2025-05-05:
The current implementation of multiple, consecutive I2C reads in one transaction skips the first byte of every continued read.

At least on my hardware here (SAMD51P), the first byte gets filled with the last byte of the previous read. For example, if an ascending sequence of bytes is read into multiple buffers, the following will actually be written into the buffers:

```text
1st read op: [00, 01, 02, 03, 04]
2nd read op: [04, 05, 06, 07]
3rd read op: [07, 08, 09, 10, 11, 12, 13]
4th read op: [13, 14, 15]
```

This bug seems to have been introduced with the new ACK handling in [`fill_buffer`](https://github.com/atsamd-rs/atsamd/blob/master/hal/src/sercom/i2c/reg.rs#L431). After having read the complete buffer, a NACK is scheduled [here](https://github.com/atsamd-rs/atsamd/blob/master/hal/src/sercom/i2c/reg.rs#L449) to be sent later. However, if the next action is another read, [`cmd_read`](https://github.com/atsamd-rs/atsamd/blob/master/hal/src/sercom/i2c/reg.rs#L380) does not get called in between, thus causing a byte to be read twice.

In a short test, I injected a call to `cmd_read` into the [`continue_read`](https://github.com/atsamd-rs/atsamd/blob/master/hal/src/sercom/i2c/reg.rs#L480) method right before the call to `fill_buffer`. This seems to work for me.

fko-kuptec, 2025-05-05:
Wow, that was faster than I could provide the pull request xD

Sympatron, 2025-05-06:
It was a test if Github Copilot could fix this with the description you provided. It did but your PR has better documentation.

PR #859 (2025-05-07): fix(i2c): acknowledge last byte when continuing read
Should fix #858
closed 2025-05-07 by jbeaurivage