atsamd-rs/atsamd #867: docs.rs still shows old v1 clock api

nyurik, 2025-05-09:
it seems https://docs.rs/atsamd-hal/0.21.3/atsamd_hal/clock/index.html only shows v1, but not v2 of the clock API

jbeaurivage, 2025-05-09:
@nyurik, the docs.rs version only shows docs for the SAMD21. Currently clock v2 is only implemented for SAMx5x chips. You can find the docs for every chip flavour using [this table](https://github.com/atsamd-rs/atsamd/?tab=readme-ov-file#hal---hardware-abstraction-layer).

closed 2025-05-09 by jbeaurivage

nyurik, 2025-05-10:
@jbeaurivage thx for the quick reply!  This was exactly what i was looking for - I got a bit confused by the docs.rs - probably because i didn't see anything there about it not documenting the whole crate, but just some relatively old variant - would it make sense to configure docs.rs to show the latest set of features?  Or maybe have some big warning in the main page with the same table as the readme? Note that it is easy to include the readme into the docs with something like this

```rust
#![cfg_attr(feature = "default", doc = include_str!("../README.md"))]
```

Lastly, would it make sense to generate exactly the same list of doc variants as there are PACs?  This way readme can be simplified to just have one table with PAC and doc links?  If some are duplicates, it is not that big of a deal I think.

jbeaurivage, 2025-05-12:
Good suggestion to add all the PAC docs to the main HAL readme. I think the readme does need a little cleanup, seems like some PACs are missing from the table.