atsamd-rs/atsamd #912: `clock::v2` API migration discussion and tracking

kyp44, 2025-06-27:
# Summary

This issue is meant to track and discuss issues related to migration to the clock v2 API.

The clock v2 API "...provides a simple, ergonomic, and most of all **safe** API to create and manage the clock tree in ATSAMD5x and E5x devices. It uses [type-level programming techniques](https://docs.rs/atsamd-hal/latest/atsamd_hal/typelevel/index.html) to prevent users from creating invalid or unsound clocking configurations."

The key feature compared to the [clock v1 API](https://docs.rs/atsamd-hal/latest/atsamd_hal/clock/index.html) is that the clock tree configuration is validated at compile time using type-level programming, which is better for most use cases in which the clock tree is set up once at the beginning of the program and then not changed.

# Status

Currently, the v2 API is only implemented for `thumbv7` targets, and is not widely supported throughout the HAL, making it currently largely unusable for most projects.

# Tasks

The following are pending identified tasks at the highest level:

1. Add clock v2 API support throughout the entire HAL for `thumbv7` targets.
2. Port the clock v2 API to `thumbv6m` targets.
3. Add clock v2 API support throughout the entire HAL for `thumbv6m` targets.

# Task 1: HAL support for `thumbv7` targets

Discussion is needed on the approach to take on this, but it should be uniform across the HAL for all chip variants. Some peripherals do already have support, but the way the support is implmeented does not seem to be consistent. Some discussion points, that will be converted to implementation guidelines as they are resolved:

1. Initially, should both clock v1 and v2 be supported for all peripherals?
2. Should the user select which API version to use via a cargo feature, causing the user-facing peripheral constructors to be different?
4. If not, should v2 support be added as an alternate constructor to preserve backward compatibility (as was done in [`Delay`](https://github.com/atsamd-rs/atsamd/pull/803)), or should the primary constructor be changed to use v2 and an alternate v1 constructor created, which breaks backward compatibility but would better faciliate eventually full migration as the v1 constructors can be deprecated.

Here are various PRs and issues related to this:

- PR #803 (Merged) adds alternate support to [`Delay`](https://docs.rs/atsamd-hal/latest/atsamd_hal/delay/struct.Delay.html).
- PR #814 (Merged) adds support for the [`Adc`](https://docs.rs/atsamd-hal/latest/atsamd_hal/adc/index.html), which seems to be mandatory.
- Issue #642 discusses that the RTC can be used without configuring the clock, which is a known issue that was discussed further in PR #845.

# Task 2: Port the API to `thumbv6m` targets.

Work on this is ongoing:

- PR #892 (Draft)
- [A fork](https://github.com/VitalBio/atsamd/tree/vital-updates) that does this. It has recent commits, but is way behind master.

# Task 3: HAL support for `thumbv6` targets

This of course depends on the completion of Task 2.

rnd-ash, 2025-06-28:
Having been implemented a few peripherals (ADC (#814)/DAC (#904)/EVSYS (#882)) recently, I think I can say for sure that we should consider a consistent way of handling the PCLKs and AHBclks.

For some peripherals, the constructor takes an `&Pclk`, which means there is no guarantee that clock will stay enabled or stay at the same frequency whilst the Peripheral is active. I think we should therefore make it such that each peripheral takes ownership of the PCLKs and AHBclocks. Then each peripheral has a `free` method that returns these clocks, guaranteeing that the clocks can only be modified with the peripheral not active.

That said, In #850, I noticed that the EIC (And some other peripherals), have the ability to choose between OSC32K and GCLKs as their clock source (OSC32K can run in sleep but provides a slow ticker, GCLK can only run with the processor awake but has a much higher speed). This means we should also provide `swap_clock` methods to these kind of peripherals such that the clock source of the peripheral can be switched (I believe the PCLK has a SourceID which we can enforce at compile time depending on the requirements of the peripheral). In this case, we should make it such that the PCLKID does not add to more Generic Pollution of the peripheral itself.

kyp44, 2025-06-28:
@ianrrees Yes, maybe the effort should be on just migrating peripherals fully to v2 instead of worrying about how to best provide compatibility with both.

@rnd-ash Thanks for the additional insights and bringing up the awareness of the new peripherals that you've added. I like the idea that all peripherals should have a `free` method that consumes the peripheral struct and returns anything that was required to use the peripheral (e.g. the `pac` and the clock tokens).

What I will do here soon (unless someone beats me to it) is to assess all currently implemented peripherals (and upcoming ones if there are open PRs) to see whether they support v1 and/or v2, and summarize the results in a table. This will help us identify what all needs to be done. For now the focus will be on Task 1, so just restricted to the `thumbv7` targets.

supersimple33, 2025-06-30:
Also, a similar thing happens when trying to use timers all together via clock::v2 since the following lines have an obvious error since mclk has been moved when trying to create the timer.

```
let (mut buses, clocks, tokens) = clock_system_at_reset(
        peripherals.oscctrl,
        peripherals.osc32kctrl,
        peripherals.gclk,
        peripherals.mclk,
        &mut peripherals.nvmctrl,
    );
let (tc2_3, gclk0) = Pclk::enable(tokens.pclks.tc2_tc3, clocks.gclk0);
let mut timer = TimerCounter::tc3_(&tc2_3.into(), peripherals.tc3, &mut peripherals.mclk);
```

kyp44, 2025-07-02:
@supersimple33 Yes this is the fundamental problem with v2 support being only partially implemented in the HAL.

kyp44, 2025-07-04:
Just added a table above of all peripherals and their status with respect to v2 migration. There is a lot of low hanging fruit here, and I will probably start issuing PRs.

@rnd-ash I have a question. In your update of the ADC and addition of DAC, you require ownership of the `ApbClk`, but only a reference to the `Pclk`, which is not retained. The other v2 peripheral, the CAN, _does_ require full ownership of its `Pclk`. In the "1:1 clocks" section of the `clock::v2` module documentation, it is implied that ownership transfer is required to ensure that the PCLK is not disabled or modified while in use by the peripheral. What is the reason for taking a reference instead?

kyp44, 2025-07-04:
@rnd-ash Gotcha, it seems to me like not taking full ownership of the `Pclk` structs is a mistake.

Can anyone (@jbeaurivage @ianrrees  @bradleyharden ) confirm this?

ianrrees, 2025-07-04:
I reckon the peripheral should own the `Pclk`.

At some stage, we're going to need to think about sleep modes etc, and in that context being able to safely adjust clock trees will be important.  I imagine this sort of ownership model will be quite helpful in that.

ianrrees, 2025-07-05:
Keeping PRs easy to review and maintain is nice, but whatever works.  I guess for something like deciding on an approach to use, it might be better to just have one representative peripheral (less friction to change as required).  But if it's not a lot of change per peripheral then I've got no problem with doing the same thing to multiple peripherals in one PR.

kyp44, 2025-07-08:
@rnd-ash The GCLK cannot be disabled as long as the `Pclk` exists. The reason is that enabling the `Pclk` will increment the compile-time consumer counter (the `N` generic type) for the `EnabledGclk`, and the `disable` method is only implemented when the consumer counter is zero, i.e. when it is [`U0`](https://docs.rs/typenum/latest/typenum/consts/type.U0.html). Users cannot increment or decrement this consumer counter manually with safe Rust. Refer to the **Tracking N at compile-time for 1:N clocks** section of the `clock::v2` module documentation for details about how these counters work.

kyp44, 2025-07-14:
A couple of larger issues with this effort need to be discussed:

1. I tried to migrate the RTC and its RTIC monotonic, and ran into the same issue that the ADC has, as noted above. The issue is that, to use the v2 API correctly, peripherals requiring a `Pclk` need to own it, and the `Pclk` can have different sources via its `I: PclkSourceId` generic. This means that the peripheral struct itself (e.g. [`Adc`](https://docs.rs/atsamd-hal/latest/atsamd_hal/adc/struct.Adc.html)) needs to have the same generic parameter. The `RtcOsc` is analogous with its `I: RtcSourceId` generic. The problem is that, on `thumbv6` targets, this generic on the peripheral struct and its `impl` blocks cannot exist (since they currently have no v2 API), leading to requiring separate `impl` blocks for each chip family, hence potentially duplicated code. This is explained nicely in a [comment for the `Adc`](https://github.com/atsamd-rs/atsamd/blob/bfdfcb949ada0c59e52641cd50e62001de90b9af/hal/src/peripherals/adc/mod.rs#L212-L219). There are two approaches to solving this problem:
    1. The ADC has a temporary solution of requiring only a `&Pclk` to construct it, ensuring that the PCLK is enabled at construction time, but allowing it to be disabled later, potentially while the peripheral is still in use. It may or may not be worth implementing this stop-gap solution for other applicable peripherals, though it does make converstion to full ownership of the `Pclk` relatively easy once the v2 API _is_ available on `thumbv6` targets.
    2. Perhaps it would be better to put effort into fast-tracking the porting of the v2 API to `thumbv6` targets as started in PR #892, before migrating applicable peripherals.

2. There is another issue with broken BSPs or BSP examples when more than one peripheral is used but their migrations are being done in separate PRs. A good example of this is the `nvm_dsu` example in the `feather_m4` BSP, which uses both the DSU and the USB peripherals. The DSU is migrated in PR #925, and the USB in PR #916. The DSU migration in particular has no choice but to leave this example broken as mentioned in PR #925. I'm not sure what the solution should be to this dilemma, but fixing some examples may need to be its own PR once the peripheral migration PRs are both merged.

rnd-ash, 2025-07-18:
@kyp44 I think 926 is ready to go, then we will have QSPI supporting the Clock V2 API (With some added features)

ianrrees, 2025-07-29:
Cheers, thanks for the info @kyp44 !  Definitely seems like a project in its own right, and for after the thumbv6 work is merged.

kyp44, 2025-11-13:
@rnd-ash That sounds good. I have not had a lot of time to work on this stuff lately, unfortunately, but I am still working on an easy-to-use USB serial crate. After that's released, I'll move back to migrating the `thumbv7` peripherals that remain.