atsamd-rs/atsamd #936: SAMD51J PAC is missing TC6

ianrrees, 2025-08-29:
I'm at a Rust conference, and have spent some time with @ahdinosaur on the Feather M4 neopixel situation - trying to use the TC peripheral to generate the bit stream for it.  The first snag we encountered is that the TC6 peripheral isn't available due to an apparent error in the SVD, which is the same as the version from https://packs.download.microchip.com/

rnd-ash, 2025-10-15:
@ianrrees are you sure? SAMD51J has 6 TCs (Numbered 0-5), only the ones with 8 timers have TC6+7

[image]

ianrrees, 2025-10-17:
Oh that's interesting - I'd looked at the pin MUX table and seen PB03 goes to TC6/WO[1], and just hacked in some local changes to the PAC to make it work.  The peripheral does seem to work (I was having some trouble getting the WS2812 protocol to work just right, but I think that was more down to timing constraints being hard to meet) but it does seem like that's actually off-label...  Thanks for figuring this out, I'll close the issue and update #752

closed 2025-10-17 by ianrrees